softwarecrash/Daly2MQTT #165: [Bug]: ESP01 after reboot  does not connect at all or poorly to the AP

Jirka10, 2024-03-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

ESP-01S (1MB Flash)

### What happened?

ESP won't connect to ASUS router after entering SSID and PWD and restarting. Sometimes it takes up to 30 minutes. I think a permanent setting to the B/G standard would help. I had a similar problem with EspEasy, where after entering force B/G the problem went away.
Thanks for the help J.

### Screenshots / Fotos

No

### Version

2.x.x and above

### Relevant livejson output

```json
No
```

### Which BMS is connected?

BMS 16S 100A

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-03-04:
hello,
i don't know an option to force the esp to the b/g network mode.
there is a known bug with "PMF" when your router have this option, disable it.

Jirka10, 2024-03-06:
They write about the option of forced mode here : https://github.com/letscontrolit/ESPEasy/issues/2012 or here : https://www.reddit.com/r/esp8266/comments/5inhtw/how_to_change_radio_type_from_80211g_to_80211n/ or here: https://github.com/sidoh/esp8266_milight_hub/issues/448  Unfortunately, I can't implement this into the source code and then compile it.
J.

softwarecrash, 2024-03-06:
seems like a problem with your asus router (asus make crap firmware)
fritzbox, unify, and much other works without any problems.
i cant make a mixed b/g function

try this
[Daly2MQTT_d1_mini_2.13.0A1_debug.zip](https://github.com/softwarecrash/Daly2MQTT/files/14516592/Daly2MQTT_d1_mini_2.13.0A1_debug.zip)
but i cant build in it actualy as option

Jirka10, 2024-03-08:
Hi, this software version works absolutely perfectly, wemos connects immediately after reboot.
Good work .
Thank you

closed 2024-04-22 by github-actions[bot]