softwarecrash/Daly2MQTT #169: SOC is set to the status of Pack_SOC after a reboot!

AlfredSch, 2024-04-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Hello, I am using Daly2Mqtt in conjunction with IoBroker. My problem is that when the Daly2Mqtt reboots or logs into the W-Lan again, it automatically writes the SoC of the BMS with the value that was last set in the IoBroker ( Pack_SOC ). How can I prevent this so that the Pack_SOC is only written when I want it to be.
Greetings & thanks
Fred

### Screenshots / Fotos

no

### Steps To Reproduce

Hello, I am using Daly2Mqtt in conjunction with IoBroker. My problem is that when the Daly2Mqtt reboots or logs into the W-Lan again, it automatically writes the SoC of the BMS with the value that was last set in the IoBroker ( Pack_SOC ). How can I prevent this so that the Pack_SOC is only written when I want it to be.

### Version

2.x.x and above

### Relevant livejson output

```json
no
```

### Which BMS is connected?

Daly BMS LIFEPO4 16S 48V UART

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-04-10:
hello,
please try to set the settings of the mqtt server to this
[image]

AlfredSch, 2024-04-10:
Hello and thanks for the quick reply :-)
I have seen today that there is an update, with this version the error is gone. I have also adjusted the setting of the MQTT instance
Greetings
Fred

closed 2024-04-10 by softwarecrash