softwarecrash/Daly2MQTT #171: [FEATURE] Add possibility to change DfilterCur

erxbout, 2024-04-11:
### Is your feature request related to a problem? Please describe

I bought a battery containing a DALY BMS and I would like to change the parameter DfilterCur to a smaller value..
If the BMS is measuring a current that is below that value then its considered 0  (it seems like its configured to be 2A and thats too much current as a border for me because I have smaller currents than that)

The problem is that the SOC becomes useless pretty fast as the BMS considers a current of for example 1A as 0A and does not change the SOC accordingly..

It seems like the parameter can be changed with a tool called Sinnowealth BMS Tool but needs some hardware for it that I do not have..
See https://www.akkudoktor.net/forum/bms-recbms/daly-bms-soc-verhalten/ for a small discussion on this..
Screenshot of the answer here:
[image]

But maybe this can be build into this project?

### Describe the solution you'd like

A possibility to change said parameter

### Describe alternatives you've considered

Alternative for me is buying a SmartShunt and use the readings from there..

### Additional context

The Daly2MQTT board works as expected, thank you very much for your work but sadly the SOC is not quite useful under the stated conditions..

softwarecrash, 2024-04-11:
hello,
thats  not possible, but you can use a uart-usb adapter and get the daly pc app to adjust this value, but remember when you set it too smal, its bossible that the shunt overload and burn out.

so the second thing why not is that all these values are verry critical, and with the wrong values you battery can burn. this is why im not implement any of this functions. think about that someone get into your network and find out the values... your house burn good ;)

so the soc set itself to 100% back when you hit the cell charge limit

closed 2024-04-11 by softwarecrash

erxbout, 2024-04-11:
Understandable, thanks for the quick reply..

Yea my problem is that SOC 100% is reached while the voltage is far from cell charge limit..
Because of the discharge current being too small most of the time (during night time standby consumption of household)
The reset occurs many hours of charging later and that bothers me..

But if something can burn just by editing this value I am not going to consider this further..
My goal would have been 0.1A since the SmartShunt i was looking for could do that..

I just read information about small currents that appear when cables are close together and can get measured as well from the bms messing up SOC again so there is a limit in place where current is considered..
Nobody mentioned hardware limitations but there is always something to learn..

Thanks again!