softwarecrash/Daly2MQTT #178: [Bug]: Lost Wifi Connection

schpuntik, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

After 3-5 days the wifi connection is lost. The LED flashes 4 times. After the reset it reconnects to the network and works again for 3-5 days.

### Screenshots / Fotos

[image]

### Steps To Reproduce

By time

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Powerpack2_200","IP":"192.168.178.153","ESP_VCC":3.355,"Wifi_RSSI":-50,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.14.0","Flash_Size":4194304,"Sketch_Size":456528,"Free_Sketch_Space":3715072},"Pack":{"Voltage":53.1,"Current":-3.3,"Power":-175.23,"SOC":76.2,"Remaining_Ah":152.4,"Remaining_kWh":8.09244,"Cycles":39,"BMS_Temp":24,"Cell_Temp":24,"cell_hVt":3.6,"cell_lVt":2.6,"cell_hVt2":3.65,"cell_lVt2":2.5,"pack_hVt":55.8,"pack_lVt":46.4,"pack_hVt2":58.4,"pack_lVt2":44.8,"High_CellNr":2,"High_CellV":3.324,"Low_CellNr":1,"Low_CellV":3.323,"Cell_Diff":1,"DischargeFET":true,"ChargeFET":true,"Status":"Discharge","Cells":16,"Heartbeat":212,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":3.323,"CellV_2":3.324,"CellV_3":3.324,"CellV_4":3.324,"CellV_5":3.324,"CellV_6":3.324,"CellV_7":3.324,"CellV_8":3.324,"CellV_9":3.324,"CellV_10":3.324,"CellV_11":3.324,"CellV_12":3.324,"CellV_13":3.324,"CellV_14":3.324,"CellV_15":3.324,"CellV_16":3.324},"CellTemp":{"Cell_Temp_1":24}}
```

### Which BMS is connected?

SmartBMS 16S 200A

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2024-06-10:
Whats your Power source and Router?

schpuntik, 2024-06-10:
Router: FritzBox 7590
Power: Step-Down  Module  1,5A
Daly2MQTT has been running smoothly for me since 2.10.0
The problem only appeared after the update to 2.14.0

softwarecrash, 2024-06-10:
That Sounds wired, it have No signifikant Changes in the WiFi part.
Try Turn Off the PMF in the Fritzbox, under WiFi Security. ESP have Problems with it.
Can you See the Log of the Fritzbox what Happen with the ESP as Client?

schpuntik, 2024-06-10:
PMF is switched off. Unfortunately I can't see any log from the ESP. But I have 2 ESPs nearby on the same connection and they work without any problems.

schpuntik, 2024-06-10:
I'm thinking about downgrading. Back to 2.13.0

softwarecrash, 2024-06-10:
Okay, can you please Test step by step downward the releases that WE can find with wich Version the Problem comes up

schpuntik, 2024-06-10:
I have 2 batteries with Daly2MQTT. I have now flashed one of them to 2.13.0. I'll let you know if it happens again.

softwarecrash, 2024-06-10:
Thank you.
What Software Version your Fritzbox have?

schpuntik, 2024-06-10:
FRITZ!OS: 7.57

closed 2024-07-26 by github-actions[bot]

Emmpunkt, 2025-04-20:
I´m facing the same Issue.
But my Wifi Hardware is all Ubiquiti.
The ESP is a Wemos.
It starts with 2.14.0. It might take a while, several hours, till you see the ESP lossing connection. But over time its getting worse.
It looks like 2.13.0 works fine, but i´m still testing....

softwarecrash, 2025-04-20:
> I´m facing the same Issue. But my Wifi Hardware is all Ubiquiti. The ESP is a Wemos. It starts with 2.14.0. It might take a while, several hours, till you see the ESP lossing connection. But over time its getting worse. It looks like 2.13.0 works fine, but i´m still testing....

when it loose connection, is the esp starting the AP mode?

Emmpunkt, 2025-04-20:
Yes, but i was not able to connect. It seem like it is changing between the modes to fast, or is restarting/resetting.

softwarecrash, 2025-04-20:
> Yes, but i was not able to connect. It seem like it is changing between the modes to fast, or is restarting/resetting. I unplugged the BMS to flash it.(i have seperate 5V, not from the BMS)

sounds crazy, but cant reproduce it here at the moment