softwarecrash/Daly2MQTT #180: [HELP] loosing SOC when permanently charged

TheOriginal92, 2024-07-20:
### A clear and concise description of what the problem is.

Hey,

im not sure if this is an issue or wanted behavior. But when I keep my bms (HIBMS) charged at 12,8V the SOC keeps dropping till the pack is reported as empty.

Seems like a calculated drain with a time base. How can this be changed?

Thank you very much!

softwarecrash, 2024-07-21:
hello,
that  is calculated, so when the bms is not in  deepsleep it calculate the self energy drop and decrease the soc.
there is nothing changeable in this behavior, only a short wake time

TheOriginal92, 2024-07-25:
@softwarecrash thank you for this explanation. How or where can I set a short wake time?

softwarecrash, 2024-07-25:
> @softwarecrash thank you for this explanation. How or where can I set a short wake time?

in the app for the bms, search under settings for wake time or similar

closed 2024-09-08 by github-actions[bot]