softwarecrash/Daly2MQTT #185: [HELP] <title> Gemeinsamer Code zum Auslesen eines MPTT Inverters und dem Daly2MQTT

ChrisT7912, 2024-09-28:
### A clear and concise description of what the problem is.

Hallo,
ich hoffe ich bin an dieser Stelle richtig und würde gern einmal meine Problemstellung kurz erklären. Ich habe einen eSmart3 kompatiblen MPPT Ladegeichrichter (eco Worthy branded ipandee Explorer NS), den ich derzeit über einen leicht erweiterten [Code](https://github.com/joba-1/Joba_ESmart3/tree/master/examples/Monitor_ESmart3) von Joba-1 Esmart 3  Monitor mit einem angeschlossenen 485 Konverter und softwareserial auslese und die Daten in den iobroker überführe. Das klappt soweit gut. 

Für die Batterie habe ich ein Daly 4S 100A (altes Modell) BMS und den BT Adapter dazu. Nun würde ich gern die Daten auch per MQTT auslesen und ebenfalls für meine Steuerlogik im iobroker verwenden.

Besteht denn die Möglichkeit, dass Du / Ihr den Code ergänzt und in Eurer Webinterface mit integriert? Ich habe davon leider überhaupt keinen Plan :(

Gruss
Chris

### Additional context

Natürlich müsste das nicht umsonst erfolgen. Ich würde für die Entwicklung natürlich auch eine finanzielle Unterstützung anbieten

softwarecrash, 2024-09-28:
hallo,
dafür arbeiten wir gerade an solar2mqtt, das soll modbus support bekommen, ob es den esmart3 unterstützen wird/kann ist unklar.
aber dein modbus, in den daly code einzustricken ist unmöglich bzw würde der esp platzen.
ansonsten musst du halt zwei esp  nutzen um von beiden geräten die daten zu bekommen

ChrisT7912, 2024-09-29:
Alles klar. Sehr schade bzw. hatte ich schon vermutet, dass es dann wohl auf 2 Geräte hinauslaufen würde. Ich werde mir dann sicherlich eine eurer fertigen Platinen bestellen, da die direkt im Batteriekoffer mit verschwinden kann.

Trotzdem danke für die Info

closed 2024-09-29 by ChrisT7912