softwarecrash/Daly2MQTT #188: [HELP] <title> Problem mit MQTT (Device_Control)

sonny1985, 2024-11-04:
### A clear and concise description of what the problem is.

Hallo . 

Hab einen Wemos im Einsatz. schon recht lange, keine Probleme damit.`Strom direkt per USB. Vom BMS kommt nur GND, D6 und D7.

Heute als ich remote (Per MQTT) den LADEFET abschalten wollte

Ich sende folgendes: 
[image]

Hab mich darum nichtmehr gekümmert, weil es so anundfürsich ohne Probleme funktioniert hat. 

Hier der auszug aus dem WebSerial, wo man sieht, das es ankommt, das BMS das zwar umschaltet, er es aber nicht tut. Der Schiebeschalter bleibt auf "true", und am MQTT seh ich auch das er noch aktiv ist. Selbiges verhalten mit dem DischargeFET.

[WebSerial] Connecting...
[WebSerial] Connected...
<BMS > CRC FAIL
<MQTT> Data sent to MQTT Server
<MQTT> MQTT Callback: message received: false
<MQTT> message recived: false
<MQTT> MQTT Callback: switching Charging mos off
<BMS > Switch charge MOSFETs to 0
<MQTT> Data sent to MQTT Server
<WEBS> Data sent to WebSocket
<WEBS> Data sent to WebSocket

Gibt es da vielleicht eine Idee ? Wäre um Hilfe dankbar :)
LG und danke vorab !

sonny1985, 2024-11-04:
Eben nochmal getestet, diesesmal klappte es per MQTT. Nach übertragen des Topics wurde es 30 sekunden später in die Tat umgesetzt. 
Ich bilde mir ein das ging immer sofort. Bin bisschen ratlos.

softwarecrash, 2024-11-05:
wenn das länger dauert ist die kommunikation mti dem bms evt ein wenig instabil.
normal sind da keine probleme bekannt bisher

sonny1985, 2024-11-05:
Hi, okay danke für die Info. Ich meld mich morgen nochmal, ich werd das heut und morgen noch 2-3 mal testen. Wenns dann klappt hätte es sich eh erledigt :).
Lg

sonny1985, 2024-11-06:
Hallo nochmals.

Ich hab das nochmals paar mal getestet, funktioniert einwandfrei nun. Sorry für den falschen Alarm hier, ich hab aber absolut keine Ahnung was da nicht gepasst hat. Wird vermutlich an einer kurzfristigen Störung der Kommunikationsleitung liegen.

OK, Das kannman hier definitiv zumachen, Danke !.
LG

softwarecrash, 2024-11-06:
sehr schön :)