softwarecrash/Daly2MQTT #194: [Bug]: Unable to install V2.14.2 from Scratch

gzi01, 2025-02-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Device did not connect to my Wifi Network
The device was not able to connect to my Wifi network.
Tried this several times with 2 different SSID  . This is my 3rd Daly2MQTT device  , I know the process.

### Screenshots / Fotos

n/a

### Steps To Reproduce

1. Connect Wemos to PC via USB
2. I flashed Release V2.14.2 , file Daly2MQTT_d1_mini_2.14.2.bin with Tasmotizer
3. connected to setup Wifi
4. enter Wifi credentials and saved 
5. Device reboots and then after a while opened its setup wifi again

Workaround
- flashed old  Daly2MQTT_d1_mini_2.10.2.bin, 
- did the Wifi setup 
- made an upgrade via Daly2MQTT_d1_mini_2.14.2_OTA.bin.gz . 
 
This seemed to work. At least the device is online for a while now.

### Version

2.x.x and above

### Relevant livejson output

```json
n/a
```

### Which BMS is connected?

none yet

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2025-02-10:
have you characters like % in your password?

gzi01, 2025-02-10:
No, no special characters

softwarecrash, 2025-02-10:
can´t reproduce the issue here, some possible problems are mixed wifi, the 2.13 have a temporarly fix for 11g network issue

closed 2025-03-27 by github-actions[bot]