softwarecrash/Daly2MQTT #202: [Bug]: Spannung Zelle 1 !

sonny1985, 2025-08-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Used Hardware?

Other (please write your hardware in 'What happened?')

### What happened?

Ich hab 2 Dieser Geräte im Einsatz (2 16Zeller), allerdings auf einem NodeMCU Board, nicht dem D1 Mini :)

Ich hab folgendes Problem, das mir das BMS falsche Spannungen misst wenn der ESP an der UART angeschlossen ist.

Ich beobachte das nun schon recht lange. Da ich im moment auf ein Victron System umbau, ist es mir aufgefallen. Die Zellenspannungen entsprechen NICHT dem was mir "übermittelt" wird.

Auch die Zellendifferenz nicht. Beide Speicher haben 0,002V Differenz. Auffällig ist, das die Werte der ersten Zelle (Beide Speicher ) Mit dem übereinstimmen, was ich selber messe. und alle anderen zu "hoch" dargestellt werden.

Ich hab nur RX, TX und GND angeschlossen, versorgt wird er extern über ein USB Netzteil.
Ich hab ein neues Daly und ein altes BMS. Das neue hat 2 UART Ausgänge. Wo ich schön mit dem BT dongle von Daly beobachten kann... Steck ich den Daly2MQTT ab, nach 2 Sekunden stimmen alle Zellenspannungen wieder und auch die Differenz ist weg. 

Kann mir bitte jemand erklären was ich da falsch mache ? Ich hab BEIDE Platinen bereits erneuert (Verdacht auf defekt) und es zeigen BEIDE , bei BEIDEN Speichern , das gleiche verhalten ...

Differenz was mir "übermittelt" wird, 19-20 mV (Tatsächlich sind es aber 2 mV. ) .

Wie gesagt, steck ich eine Platine beispielweise ab: Passt alles , so wie es sein soll. Steck ich sie wieder an, Verändert da irgendetwas die Zellspannungen und sorgt für Terror im BMS .

### Screenshots / Fotos

[image]

### Version

2.x.x and above

### Relevant livejson output

```json
{"Device":{"Name":"Daly2MQTT2","IP":"192.168.0.133","Flash_Size":4194304,"Sketch_Size":463216,"Free_Sketch_Space":3706880,"CPU_Frequency":80,"Real_Flash_Size":4194304,"ESP_VCC":3.27,"Wifi_RSSI":-53,"Relais_Active":false,"Relais_Manual":false,"sw_version":"2.15.0","Free_Heap":29880,"json_space":1296,"runtime":1261,"ws_clients":0,"MQTT_Json":false},"Pack":{"Voltage":53.8,"Current":10.8,"Power":581.04,"SOC":97.9,"Remaining_Ah":308.3,"Remaining_kWh":16.58654,"Cycles":3,"BMS_Temp":23,"Cell_Temp":23,"cell_hVt":3.6,"cell_lVt":3,"cell_hVt2":3.65,"cell_lVt2":2.9,"pack_hVt":57.6,"pack_lVt":48,"pack_hVt2":58.4,"pack_lVt2":46.4,"High_CellNr":5,"High_CellV":3.37,"Low_CellNr":1,"Low_CellV":3.35,"Cell_Diff":20,"DischargeFET":true,"ChargeFET":true,"Status":"Charge","Cells":16,"Heartbeat":122,"Balance_Active":false,"Fail_Codes":""},"CellV":{"CellV_1":3.35,"CellV_2":3.369,"CellV_3":3.369,"CellV_4":3.369,"CellV_5":3.37,"CellV_6":3.37,"CellV_7":3.37,"CellV_8":3.369,"CellV_9":3.37,"CellV_10":3.369,"CellV_11":3.37,"CellV_12":3.369,"CellV_13":3.369,"CellV_14":3.37,"CellV_15":3.369,"CellV_16":3.367},"CellTemp":{"Cell_Temp_1":23,"Cell_Temp_2":23}}
```

### Which BMS is connected?

2 mal UART, einmal alte Version und einmal das neue.

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

sonny1985, 2025-08-08:
Was ich probieren kann : Den D1 Mini- Vielleicht liegt es daran das ich die NodeMCU´s verwende .
Kann es daran liegen ?

sonny1985, 2025-08-08:
So ich hab eine V5 Platine mal gekauft vor einem halben jahr. Diese nun mit einem D1 Mini montiert testhalber.

Zelle 1. + / - 0,020 V. die springt permanent hin und her. (3,485v 5 sekunden später 3,504 2-3 publishs weiter wieder 3,485 V) Stimmen tun im übrigen 3,504. Das heisst er zeigt mir bei Zelle 1 "zuwenig" an.

Damit einhergehend springt auch die Zellendifferenz). Das war im übrigen auch der Grund warum ich die V5 entfernt hab, und NodeMCUs in verwendung hatte. 

Ich bin ratlos. -.-

sonny1985, 2025-08-08:
Nun habe ich das V5 wieder entfernt, und das NodeMCU angeschlossen. Nun spinnt das andere Daly2MQTT auch wieder....
(hellblau- Zelle 1)
Die hängen doch nicht zusammen. Lediglich beide Speicher sind per Kreuzschaltung parallel verbunden. Ich nehm von den BMS aber keine Spannung ab (also nur GND RX und TX am Node MCU, per USB vom extra Netzteil mit strom versorgt)

[image]

Das V5 war OHNE USB, dafür direkt am BMS 

Ich blick da nichtmehr durch.

sonny1985, 2025-08-08:
So ich muss das nun auflösen. Ich bin durch Zufall dahintergekommen:
Es ist das USB Netzteil gewesen 

[image]

Die Problematik daran war, das es zwar anständig Strom lieferte. Ich allerdings mit einer 5 Meter verlängerung zu EINEM der Dalymqtt´s gearbeitet hab, wo auch noch ein Raspberry mitlief. Es kam zu einem Spannungsabfall. 
Ich hab da draussen 2 Raspberrys 4, in einem Argonone Case mit nvme am laufen. Dort gibt es ein Rotes (Ist eingeschalten) und ein grünes Lämpchen drauf (das Grüne ist für Datenträgeraktivität). An einem arbeitet mein Iobroker, auf dem anderem ist VenusOs drauf. Der wo das Venus OS drauf ist, lief zwar, aber es leuchtete die Rote lampe nicht. (zumindest erlosch sie nach einer gewissen Zeit)
Intuitiv Teste ich dann ein anderes 3A USB Ding, in näherer Umgebung wo ich auf diese Verlängerung verzichten konnte und voila , alle Probleme waren behoben. 

Ich weiss nicht wieviele ESPs ich ausgetauscht hab deswegen, weil es ja immer ne Zeit lang funktionierte. Ich hab simpel gesagt den Spannungsabfall in dieser elendig langen 5V Leitung nicht beachtet.

Echt- man sieht vor lauter Bäume oft den Wald nichtmehr.
Ich schmeiss ne Kaffeerunde !. 
Das hier kann man zumachen. und Asche auf mein Haupt.

softwarecrash, 2025-08-08:
> So ich muss das nun auflösen. Ich bin durch Zufall dahintergekommen: (Ich hab die vorigen Postings gelöscht) Es ist das USB Netzteil gewesen
> 
> [image]
> 
> Die Problematik daran war, das es zwar anständig Strom lieferte. Ich allerdings mit einer 5 Meter verlängerung zu EINEM der Dalymqtt´s gearbeitet hab, wo auch noch ein Raspberry mitlief. Es kam zu einem Spannungsabfall. Ich hab da draussen 2 Raspberrys 4, in einem Argonone Case mit nvme am laufen. Dort gibt es ein Rotes (Ist eingeschalten) und ein grünes Lämpchen drauf (das Grüne ist für Datenträgeraktivität). An einem arbeitet mein Iobroker, auf dem anderem ist VenusOs drauf. Der wo das Venus OS drauf ist, lief zwar, aber es leuchtete die Rote lampe nicht. (zumindest erlosch sie nach einer gewissen Zeit) Intuitiv Teste ich dann ein anderes 3A USB Ding, in näherer Umgebung wo ich auf diese Verlängerung verzichten konnte und voila , alle Probleme waren behoben.
> 
> Ich weiss nicht wieviele ESPs ich ausgetauscht hab deswegen, weil es ja immer ne Zeit lang funktionierte. Ich hab simpel gesagt den Spannungsabfall in dieser elendig langen 5V Leitung nicht beachtet.
> 
> Echt- man sieht vor lauter Bäume oft den Wald nichtmehr. Ich schmeiss ne Kaffeerunde !. Das hier kann man zumachen. und Asche auf mein Haupt.
> 
> Für alle definitiv : Falls das Ding funktioniert, aber man der Meinung ist : es macht nur Blödsinn- Prüft die SPANNUNG (An einem 2-3 A Ding hängt eh jeder seine Teile)- Wird die Leitung zu lang wirds kritisch :)

okay, das sind so gegebenheiten die kann man wenn überhaupt nur erahnen.
5meter ist natürlich schon ne hausnummer, hat der esp bzw die firmware da nicht gemeckert wegen spannungsabfall? (so eine meldung ist eingebaut)

softwarecrash, 2025-08-08:
na hauptsache geht, dann kann das ja zu :)

closed 2025-08-09 by softwarecrash